YanjieZe/GMR #2: Retargeting from SMPL-X(AMASS) to Robot failed

Edward7Zhang, 2025-08-05:
Thanks for ur work 
following readme steps
```
Retarget a single motion:

# single motion
python scripts/smplx_to_robot.py --smplx_file <path_to_smplx_data> --robot <path_to_robot_data> --save_path <path_to_save_robot_data.pkl>`
```
but i got a error seems like not fully work with AMASS Motion Data. 
Plz give some suggest.

[image]

YanjieZe, 2025-08-05:
Could you try this?

after install SMPLX, change `ext` in `smplx/body_models.py` from `npz` to `pkl` if you are using SMPL-X pkl files.

Edward7Zhang, 2025-08-05:
> Could you try this?
> 
> after install SMPLX, change `ext` in `smplx/body_models.py` from `npz` to `pkl` if you are using SMPL-X pkl files.

Thanks a lot, solved

[image]

nlnlz, 2025-08-20:
Hi, how ti change ext in smplx/body_models.py from npz to pkl if you are using SMPL-X pkl files.  

def __init__(
        self, model_path: str,
        kid_template_path: str = '',
        num_expression_coeffs: int = 10,
        create_expression: bool = True,
        expression: Optional[Tensor] = None,
        create_jaw_pose: bool = True,
        jaw_pose: Optional[Tensor] = None,
        create_leye_pose: bool = True,
        leye_pose: Optional[Tensor] = None,
        create_reye_pose=True,
        reye_pose: Optional[Tensor] = None,
        use_face_contour: bool = False,
        batch_size: int = 1,
        gender: str = 'neutral',
        age: str = 'adult',
        dtype=torch.float32,
        ext: str = 'npz',
        **kwargs
    ) -> None:

or

if ext == 'pkl':
                with open(smplh_path, 'rb') as smplh_file:
                    model_data = pickle.load(smplh_file, encoding='latin1')
            elif ext == 'npz':
                model_data = np.load(smplh_path, allow_pickle=True)
            else:
                raise ValueError('Unknown extension: {}'.format(ext))
            data_struct = Struct(**model_data)