YanjieZe/GMR #9: new robot

zhainiuzhanshi, 2025-08-06:
Could you provide an idea on how to adapt it to my own robot?

zhainiuzhanshi, 2025-08-08:
For the same robot, when using your own BVH file with rotation order Zrotation Xrotation Yrotation (ZXY), while LaFAN1 uses Zrotation Yrotation Xrotation (ZYX), does the ik_table need to be modified?

YanjieZe, 2025-08-08:
Yes, if it does not work you need to adjust the rotations. We currently only test LAFAN1 files yet; let us know if there are other general motion datasets that we can support. Thank you!

YanjieZe, 2025-08-08:
You can disable the joint matching in config one by one and see which rotation is not aligned; and find the rotation for it.

zhainiuzhanshi, 2025-08-08:
Just removing ik_config will cause an error, right?