YanjieZe/GMR #16: Retargeting from SMPLX(AMASS) to Robot failed

Amafde, 2025-08-08:
[image]

Thanks for your work,I follow the step of readme

```
Retarget a single motion:
# single motion
python scripts/smplx_to_robot.py --smplx_file <path_to_smplx_data> --robot <path_to_robot_data> --save_path <path_to_save_robot_data.pkl>`
```

The error is below:

```
(/workspaces/ws_rl/conda/gmr) nubot@:src/GMR$ python scripts/smplx_to_robot.py --smplx_file ACCAD/Male1Walking_c3d/"Walk B13 - Walk turn right 45_poses.npz" --save_path ACCAD/Male1Walking_c3d/"Walk B13 - Walk turn right 45_poses.pkl"
Traceback (most recent call last):
  File "/workspaces/ws_rl/src/GMR/scripts/smplx_to_robot.py", line 66, in <module>
    smplx_data, body_model, smplx_output, actual_human_height = load_smplx_file(
  File "/workspaces/ws_rl/src/GMR/general_motion_retargeting/utils/smpl.py", line 21, in load_smplx_file
    num_frames = smplx_data["pose_body"].shape[0]
  File "/workspaces/ws_rl/conda/gmr/lib/python3.10/site-packages/numpy/lib/_npyio_impl.py", line 261, in __getitem__
    raise KeyError(f"{key} is not a file in the archive")
KeyError: 'pose_body is not a file in the archive'
```

Besides, the files are below:
```
nubot@nubot-LEGION-REN7000K-26IRX:~/ws_rl/src/GMR/assets/body_models/smplx$ ls
SMPLX_FEMALE.npz  SMPLX_FEMALE.pkl  SMPLX_MALE.npz  SMPLX_MALE.pkl  SMPLX_NEUTRAL.npz  SMPLX_NEUTRAL.pkl
```
I really have no idea how to handle this,if you don't mind, give me some suggestions please.

YanjieZe, 2025-08-08:
Yes, I agree with @jaraujo98. Are you download `SMPL-X` motions from AMASS? It seems you are download SMPL or SMPL-H files.

YanjieZe, 2025-08-28:
@ZZQ0316 It is expected for these `female_stagei.npz` files or  `male_stagei.npz` files. You can add a simple if-else logic to remove  them,