YanjieZe/GMR #37: Feature Request: Support for Booster T1 23-DoF Version

Gloriabhsfer, 2025-08-25:
Hello, and thank you for this amazing project!

I was wondering if you have considered, or would be open to adding, support for the 23-DoF version of the Booster T1 ?

If this is not currently on the roadmap, could you please offer some guidance on how one might modify the existing configuration (e.g., URDF, control YAML files) to accommodate the additional degrees of freedom? Any pointers would be greatly appreciated.

Thank you for your time and for your work on this project.

YanjieZe, 2025-08-25:
Thank you for proposing. We will add.

YanjieZe, 2025-08-28:
Booster T1 23 dof has been added as your request.

closed 2025-08-28 by YanjieZe

Gloriabhsfer, 2025-09-01:
Hi, i modify the t1 23 dof for bvh config base on your smplx example. https://gist.github.com/Gloriabhsfer/d1b07693ddf226e3d83cf878e71ec85e