YanjieZe/GMR #41: Feature request: add documentation on IK config mapping table

T-K-233, 2025-08-27:
Thanks for open-sourcing this amazing library!

It would be nice to have a documentation writeup on how to write a IK config mapping table to support new robots. 

I am glad to help contribute if there is need.

YanjieZe, 2025-08-27:
Thank you for the suggestion, TK. Basically it is rotation alignment like shown here (https://github.com/YanjieZe/GMR/issues/12) ; it would be definitely good if you could help write a doc for that & pull a request! appreciate that.

bopeng5221, 2025-08-30:
I'm really looking forward to the documentation. Honestly, I don't really know what the config means. I was looking at the TWIST paper but failed to find related information. A documentation would really help!