YanjieZe/GMR #58: Runtime Error (worked before 08/28, started failing after 08/30 update)

NagareTowa, 2025-09-05:
Environment
OS: Windows 11 + WSL2 / Ubuntu
Python: 3.10 (conda env: gmr)
Installation: pip install -e . 

Data & Paths:
SMPL-X model: ~/GMR/assets/body_models/smplx/SMPLX_FEMALE.npz
Motion data (ACCAD):
~/GMR/data/ACCAD/Female1General_c3d/A15_-_skip_to_stand_stageii.npz
~/GMR/data/ACCAD/Female1General_c3d/A8_-_crouch_to_lie_stageii.npz

Reproduction Steps:
_conda activate gmr
cd ~/GMR
python scripts/smplx_to_robot.py \
  --smplx_file /home/nagaretowa/GMR/data/ACCAD/Female1General_c3d/A15_-_skip_to_stand_stageii.npz \
  --robot unitree_g1 \
  --save_path /home/nagaretowa/GMR/output/g1_A15_skip.pkl \
  --rate_limit \
  --record_video_

Using A8_-_crouch_to_lie_stageii.npz reproduces the same issue.

On 08/28 (last week) with the same machine, data, and command, it ran successfully.
After pulling updates on 08/30 (and onward), it consistently fails with the following runtime error (line numbers based on my local copy):
_File ".../scripts/smplx_to_robot.py", line 74, in <module>
  smplx_data, body_model, smplx_output, actual_human_height = load_smplx_file(...)
File ".../general_motion_retargeting/utils/smpl.py", line 2x, in load_smplx_file
  smplx_output = body_model(
File ".../site-packages/smplx/body_models.py", line 1227, in forward
  shape_components = torch.cat([betas, expression], dim=-1)
RuntimeError: Sizes of tensors must match except in dimension 1.
Expected size 511 but got size 1 for tensor number 1 in the list._

With a different motion file, I also observed:
_Expected size 31 but got size 1 for tensor number 1 in the list._

YanjieZe, 2025-09-05:
Hi, I have made sure it can work on my Ubuntu system. Could you help check what change makes such errors?

NagareTowa, 2025-09-08:
> Hi, I have made sure it can work on my Ubuntu system. Could you help check what change makes such errors?

Hello, I uninstalled and reinstalled WSL2, then reinstalled GMR, and the issue is now resolved. My guess is that pulling the latest repo on top of an older local GMR copy left some stale files and caused the problem. Thank you very much for your reply!

YanjieZe, 2025-09-08:
Good to know the issue is solved.